ansys/pyadditive #261: Examples in pyadditive v0.17.0 not working

maxcarrat90, 2024-02-15:
@pkrull-ansys Thank you very much for you hints. Now I am able to correctly run the `examples/00_additive_single_bead.py`. However, using `material = additive.material("316L")` , `machine.scan_speed = 1`, and `machine.laser_power = 195` I got zero values in all Melt Pool Dimensions, see plot below:
[image]

_Originally posted by @maxcarrat90 in https://github.com/ansys/pyadditive/discussions/256#discussioncomment-8472900_

pkrull-ansys, 2024-02-15:
@maxcarrat90 This is what I'm seeing using pyadditive v0.17.0 and Ansys 24.1 when running `00_additive_single_bead.py` and changing the following values
```
material = additive.material("316L")
machine.scan_speed = 1  # m/s
machine.laser_power = 195  # W
single-bead-316l-1-195
```
[image]

Were any other edits made to the python file? Line 120 converts from meters to microns. Did it get changed by chance? What does df.head() show (line 145)? Do you see only zeros or nan's for the values?

maxcarrat90, 2024-02-16:
@pkrull-ansys I have cloned the latest pyadditive version and reinstalled everything to be sure that the python's script are all clean. I attach the melt_pool.csv file and here is what I get on the Python 
Console at the end of the simulation:
```
C:\Users\Massimo\anaconda3\envs\pyAnsys\python.exe "C:\Program Files\ANSYS Inc\PyAnsys\PyAdditive\pyadditive-0.17\examples\00_additive_single_bead.py" 
user data path: C:\Users\Massimo\AppData\Local\Ansys Inc\pyadditive
Client 0.18.dev5, API version: 1.5.0
ServerConnectionStatus(connected=True, channel_str='127.0.0.1:64708', metadata={'Version': '24.1.0-2', 'API version': '1.4.1'})
2024-02-16 09:00:53 Connected to 127.0.0.1:64708
AdditiveMachine
laser_power: 195 W
scan_speed: 1.0 m/s
heater_temperature: 80 °C
layer_thickness: 5e-05 m
beam_diameter: 0.0001 m
starting_layer_angle: 57 °
layer_rotation_angle: 67 °
hatch_spacing: 0.0001 m
slicing_stripe_width: 0.01 m

Single input
Single bead simulation: 100%|██████████| 100/100 [01:24<00:00,  1.19it/s]

Process finished with exit code 0
```
[melt_pool.csv](https://github.com/ansys/pyadditive/files/14308624/melt_pool.csv)

pkrull-ansys, 2024-02-16:
@maxcarrat90 I'm a bit baffled. I've duplicated your configuration and run the example.
```
(.venv) 08:35 AM pyadditive $ python examples/00_additive_single_bead.py 
2024-02-16 08:36:01 Connected to 127.0.0.1:59008
user data path: C:\Users\pkrull\AppData\Local\Ansys Inc\pyadditive
Client 0.18.dev5, API version: 1.5.0
ServerConnectionStatus(connected=True, channel_str='127.0.0.1:59008', metadata={'API version': '1.4.1', 'Version': '24.1.0-2'})
AdditiveMachine
laser_power: 195 W
scan_speed: 1.0 m/s
heater_temperature: 80 °C
layer_thickness: 5e-05 m
beam_diameter: 0.0001 m
starting_layer_angle: 57 °
layer_rotation_angle: 67 °
hatch_spacing: 0.0001 m
slicing_stripe_width: 0.01 m

Single input
Single bead simulation: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 100/100 [00:43<00:00,  2.29it/s]
```

Here are the melt pool statistics I get. 
[melt_pool.csv](https://github.com/ansys/pyadditive/files/14313045/melt_pool.csv)

Questions:
1. Do you get zero values with `316L` when using a higher laser power and slower scan speed, say 700 W and 0.5 m/s?
2. Do you get non-zero values for `IN718` using laser power = 195 and scan speed = 1 m/s?

maxcarrat90, 2024-02-19:
@pkrull-ansys Thanks for your help! Regarding your questions:

1.  
[image]

2.  ```
File "C:\Program Files\ANSYS Inc\PyAnsys\PyAdditive\pyadditive-0.17\examples\00_additive_single_bead.py", line 108, in <module>
    raise Exception(summary.message)
Exception: Exception: INVALID STATE: The melt pool has become too wide and deep to produce valid results (current width = 825.000000 microns, current depth = 1200.000000 microns). We recommend you reduce the energy input by reducing Laser Power and/or increasing Scan Speed.
2024-02-16 18:14:44 Uncaught exception
Traceback (most recent call last):
  File "C:\Program Files\ANSYS Inc\PyAnsys\PyAdditive\pyadditive-0.17\examples\00_additive_single_bead.py", line 108, in <module>
    raise Exception(summary.message)
Exception: Exception: INVALID STATE: The melt pool has become too wide and deep to produce valid results (current width = 825.000000 microns, current depth = 1200.000000 microns). We recommend you reduce the energy input by reducing Laser Power and/or increasing Scan Speed.
```

pkrull-ansys, 2024-02-19:
@maxcarrat90 The server seems to be misinterpreting the parameters sent to it. On the host where the server is running, what locale is used? More specifically, what's the default character used for the decimal point? The server should account for different locales but this hasn't received much testing yet.

maxcarrat90, 2024-02-19:
@pkrull-ansys here what I get checking locale on the powershell:
```
(pyAnsys) PS C:\Program Files\ANSYS Inc\PyAnsys\PyAdditive> Get-WinSystemLocale

LCID             Name             DisplayName                                                                                                                                      
----             ----             -----------                                                                                                                                      
1033             en-US            Inglese (Stati Uniti)                                                                                                                            

(pyAnsys) PS C:\Program Files\ANSYS Inc\PyAnsys\PyAdditive> 

```

pkrull-ansys, 2024-02-19:
@maxcarrat90 It looks like your host is set up for Italian. Is that correct?

In `%TMP%/additive` there should be some folders containing a `thermal-input.json` file. Delete everything under the `additive` folder then rerun one of your simulations. Find the `thermal-input.json` file then compare the values under `MachineData` to what you used when defining the simulation.

maxcarrat90, 2024-02-20:
@pkrull-ansys Thank you very much for you support! The issue is now fixed. Now, I would like to run a parametric study using a user defined material and laser beam shape (multi-mode). Would this be possible?

pkrull-ansys, 2024-02-20:
@maxcarrat90 User defined materials are supported. Shaped lasers are not yet supported.

For user defined materials, the pyadditive documentation is a little light and refers you to documentation for our legacy application. Let me know if you need help and I'll answer your questions or put you in touch with someone who can.

**PyAdditive examples**

- https://additive.docs.pyansys.com/version/stable/examples/gallery_examples/04_using_a_custom_material.html#sphx-glr-examples-gallery-examples-04-using-a-custom-material-py
- https://additive.docs.pyansys.com/version/stable/examples/gallery_examples/10_advanced_custom_material_tuning.html#sphx-glr-examples-gallery-examples-10-advanced-custom-material-tuning-py

**Custom material further information**

- https://ansyshelp.ansys.com/account/secured?returnurl=/Views/Secured/corp/v241/en/add_print/add_UserDefMatl.html
- https://ansyshelp.ansys.com/account/secured?returnurl=/Views/Secured/corp/v241/en/add_beta/add_science_BETA_material_tuning_tool.html

maxcarrat90, 2024-02-20:
@pkrull-ansys Thank you for the links! 
Would it be possible to contribute in developing shaped laser functions? I need this feature within a research project framework and I can work on its implementation.

pkrull-ansys, 2024-02-20:
@maxcarrat90 Unfortunately, I think it would likely be used in our thermal solver, which is not open source. We do have a shaped laser feature in our backlog. If you can write up a feature request as an issue for pyadditive, I can see if what we intend meets your needs or whether we can modify it to meet your needs.

closed 2024-02-21 by maxcarrat90