ansys/pyadditive #310: User Provided Thermal Parameters should be mentioned explicitly.

safeerehman, 2024-03-19:
### 📝 Description of the feature

Currently, `ParametricStudy::generate_microstructure_permutations` takes the input `cooling_rate`, `thermal_gradient`, `melt_pool_width` and `melt_pool_depth` as arguments and passes it on to create a `MicrostructureInput`.

`MicrostructureInput` however has an additional argument `use_provided_thermal_parameters`, a `boolean` field that is set to `False` by default and `True` only if the user provides the above four arguments explicitly. 

`ParametricStudy::generate_microstructure_permutations` sets the `use_provided_thermal_parameters` internally depending upon if the arguments have a `nan` or a `None` or an actual value specified and passes it while creating the `MicrostructureInput`

This needs to be enhanced by having the user explicitly mention `use_provided_thermal_parameters` in `ParametricStudy::generate_microstructure_permutations` along with the values of each of the four arguments. 

The study data frame should have an additional column for `use_provided_thermal_parameters` and the fields for `cooling_rate`, `thermal_gradient`, `melt_pool_width` and `melt_pool_depth` should not be empty. 

### 💡 Steps for implementing the feature

- `ParametricStudy::generate_microstructure_permutations` should have an additional argument `use_provided_thermal_parameters` 
- `use_provided_thermal_parameters` is of type boolean and is `False` by default. 
- If set to `True` ensure that all `cooling_rate`, `thermal_gradient`, `melt_pool_width` and `melt_pool_depth` fields have a float value otherwise throw an exception. 
- If set to `False` these values should be the default values `MicrostructureInput` takes. 
- The study data frame should have an additional column for `use_provided_thermal_parameters` and the other four columns should always have a value in them, whether default or the user provided value.

pkrull-ansys, 2024-04-18:
@safeerehman Did this get implemented?

safeerehman, 2024-04-19:
@pkrull-ansys, not yet.