ansys/pyadditive #646: Problem with running an exmples

Matemusi, 2025-01-31:
Hello, there is some problem with additive.material("316L") (and other materials) occuring for me for some reason. Just to show it on example I ran the 01_additive_porosity.py.
And this is the output:

```
2025-01-31 18:57:06 Connected to 127.0.0.1:61200
Available material names: ['17-4PH', '316L', 'Al357', 'AlSi10Mg', 'CoCr', 'IN625', 'IN718', 'Ti64']
2025-01-31 18:57:06,774 - CRITICAL - logger - handle_exception - Uncaught exception
Traceback (most recent call last):
  File "~\Ansys\additive.py", line 58, in <module>
    material = additive.material("316L")
  File "~\ansys\additive\core\additive.py", line 444, in material
    result = self._servers[0].materials_stub.GetMaterial(request)
  File "~\grpc\_channel.py", line 1181, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "~\grpc\_channel.py", line 1006, in _end_unary_response_blocking
    raise _InactiveRpcError(state)  # pytype: disable=not-instantiable
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Exception was thrown by handler."
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-01-31T15:57:06.7501847+00:00", grpc_status:2, grpc_message:"Exception was thrown by handler."}"
>
```

I used ANSYS 2024 R1 and PyAdditive 0.17.2 (also tried others). There was a problem with DEFAULT_PRODUCT_VERSION in ~\ansys\additive\core\server_connection\constants.py, so I changed it to 241. 

Server logs are all like this:
```
2025-01-31 18:57:06.580 [INF] Microsoft.AspNetCore.Hosting.Diagnostics Request starting HTTP/2 POST http://127.0.0.1:61200/ansys.api.additive.v0.about.AboutService/About application/grpc -
2025-01-31 18:57:06.600 [INF] Microsoft.AspNetCore.Routing.EndpointMiddleware Executing endpoint '"gRPC - /ansys.api.additive.v0.about.AboutService/About"'
2025-01-31 18:57:06.636 [INF] Microsoft.AspNetCore.Routing.EndpointMiddleware Executed endpoint '"gRPC - /ansys.api.additive.v0.about.AboutService/About"'
2025-01-31 18:57:06.638 [INF] Microsoft.AspNetCore.Hosting.Diagnostics Request starting HTTP/2 POST http://127.0.0.1:61200/ansys.api.additive.v0.materials.MaterialsService/GetMaterialsList application/grpc -
2025-01-31 18:57:06.639 [INF] Microsoft.AspNetCore.Hosting.Diagnostics Request finished HTTP/2 POST http://127.0.0.1:61200/ansys.api.additive.v0.about.AboutService/About application/grpc - - 200 - application/grpc 61.3775ms
2025-01-31 18:57:06.639 [INF] Microsoft.AspNetCore.Routing.EndpointMiddleware Executing endpoint '"gRPC - /ansys.api.additive.v0.materials.MaterialsService/GetMaterialsList"'
2025-01-31 18:57:06.726 [INF] Microsoft.AspNetCore.Routing.EndpointMiddleware Executed endpoint '"gRPC - /ansys.api.additive.v0.materials.MaterialsService/GetMaterialsList"'
2025-01-31 18:57:06.726 [INF] Microsoft.AspNetCore.Hosting.Diagnostics Request finished HTTP/2 POST http://127.0.0.1:61200/ansys.api.additive.v0.materials.MaterialsService/GetMaterialsList application/grpc - - 200 - application/grpc 87.8896ms
2025-01-31 18:57:06.729 [INF] Microsoft.AspNetCore.Hosting.Diagnostics Request starting HTTP/2 POST http://127.0.0.1:61200/ansys.api.additive.v0.materials.MaterialsService/GetMaterial application/grpc -
2025-01-31 18:57:06.729 [INF] Microsoft.AspNetCore.Routing.EndpointMiddleware Executing endpoint '"gRPC - /ansys.api.additive.v0.materials.MaterialsService/GetMaterial"'
2025-01-31 18:57:06.738 [ERR] Grpc.AspNetCore.Server.ServerCallHandler Error when executing service method '"GetMaterial"'.
System.FormatException: Input string was not in a correct format.
   at System.Number.ThrowOverflowOrFormatException(ParsingStatus status, TypeCode type)
   at Additive.Server.MaterialExtension.Copy(AdditiveMaterial am, Material m)
   at Additive.Server.Services.MaterialsService.GetMaterial(GetMaterialRequest request, ServerCallContext context)
   at Grpc.Shared.Server.UnaryServerMethodInvoker`3.Invoke(HttpContext httpContext, ServerCallContext serverCallContext, TRequest request)
--- End of stack trace from previous location ---
   at Grpc.AspNetCore.Server.Internal.CallHandlers.UnaryServerCallHandler`3.HandleCallAsyncCore(HttpContext httpContext, HttpContextServerCallContext serverCallContext)
   at Grpc.AspNetCore.Server.Internal.CallHandlers.ServerCallHandlerBase`3.<HandleCallAsync>g__AwaitHandleCall|8_0(HttpContextServerCallContext serverCallContext, Method`2 method, Task handleCall)
2025-01-31 18:57:06.749 [INF] Microsoft.AspNetCore.Routing.EndpointMiddleware Executed endpoint '"gRPC - /ansys.api.additive.v0.materials.MaterialsService/GetMaterial"'
2025-01-31 18:57:06.750 [INF] Microsoft.AspNetCore.Hosting.Diagnostics Request finished HTTP/2 POST http://127.0.0.1:61200/ansys.api.additive.v0.materials.MaterialsService/GetMaterial application/grpc - - 200 0 application/grpc 20.9607ms

```

pkrull-ansys, 2025-01-31:
@Matemusi Thanks for reporting this. I'll get back to you when I've had a chance to reproduce the issue.

@deleon117 - FYSA

pkrull-ansys, 2025-01-31:
@Matemusi I'm unable to reproduce your issue using v0.17.2 of pyadditive and 2024 R1 of Ansys. After updating `DEFAULT_PRODUCT_VERSION ` to `241`, the example `01_additive_porosity.py` ran without error. My hunch is that the client and server versions that are running are not what you intend and that the versions are incompatible. To troubleshoot this, please add `additive.about()` to the example code or call it from the python command line. For example:
```
from ansys.additive.core import Additive, AdditiveMachine, PorosityInput, SimulationError

additive = Additive()
additive.about()
```

You should see output from the `about()` call similar to the following.
```
Client 0.17.2, API version: 1.4.1
ServerConnectionStatus(connected=True, channel_str='127.0.0.1:50764', metadata={'API version': '1.4.1', 'Version': '24.1.0-2'})
```

Contact me with your results and we can troubleshoot further.

Matemusi, 2025-01-31:
@pkrull-ansys  That is an output from additive.about():
```
Client 0.17.2, API version: 1.4.1
ServerConnectionStatus(connected=True, channel_str='dns:///127.0.0.1:58466', metadata={'API version': '1.4.1', 'Version': '24.1.0-2'})
```
The only difference should be in channel

pkrull-ansys, 2025-01-31:
@Matemusi What language settings are you using?

[image]

Matemusi, 2025-02-01:
@pkrull-ansys Yeah, thanks, when I changed Windows display language to English (United States) it worked. By the way, additive print features and stresses/deformations etc. export in .vtk files are not yet implemented?

pkrull-ansys, 2025-02-04:
@Matemusi I'm closing this issue. Let me know if you want it reopened.

closed 2025-02-04 by pkrull-ansys